michael-slx/weewx-weatherlink-live #1: Data Binding Issue

gary-hammer, 2020-10-21:
I've installed this driver and it is working well.
I did edit the database entries as shown in the annotated example weewx.conf.
With this entry, weewx does not start.
manager = weewx.wxmanager.WXDaySummaryManager
Only if I leave it at weewx default will weewx start.
manager = weewx.manager.DaySummaryManager
The schema is being used as a new database is created with a size of 1204 kb.

michael-slx, 2020-10-23:
Hi,

unfortunately I'm not able to reproduce your issue.
Would it be possible for you to provide a syslog/journal excerpt?

 Looking at the WeeWX code, it shouldn't really matter since both options are linked to the same thing. I'll change the example configuration anyway since `WXDaySummaryManager` is the old way of doing things and shouldn't be used anymore.

gary-hammer, 2020-10-23:
I'm sorry, I ran this on a testing server, I restore to a known state after testing, so no logs are available.
If I have other issues, I will pull logs before resetting the machine.

As this is no longer going to be given as an option, I'll close this issue.

closed 2020-10-23 by gary-hammer

michael-slx, 2020-10-23:
No problem. Thanks for your help with testing, anyway.

commit 30b4bde0 (2020-10-23): Switch to new archive service

weewx.wxmanager.WXDaySummaryManager is deprecated.

Fixes #1