michael-slx/weewx-weatherlink-live #3: Logging Success

gary-hammer, 2020-10-24:
I installed 1.0.4 and it indeed corrects the high CPU issue.
However, it introduces logging of success. This will result in very high syslog usage of course.
How can I toggle this off? I tried log_success = false in the driver stanza with no effect.

Just for your information, the next (4.2.0) version of WeeWX has some changes in the data and how it is processed. beaufort is handled differently now.

`Oct 24 10:41:56 tester weewxd[11871]: weewx[11871] INFO user.weatherlink_live.mappers: RainMapping[['1']]: First daily rain value
Oct 24 10:41:56 tester weewxd[11871]: weewx[11871] INFO user.weatherlink_live: Emitting push (broadcast) packet
Oct 24 10:41:56 tester weewxd[11871]: weewx[11871] INFO weewx.wxxtypes: Type beaufort has been deprecated. Use unit beaufort instead.
Oct 24 10:41:58 tester weewxd[11871]: weewx[11871] INFO user.weatherlink_live: Emitting push (broadcast) packet
Oct 24 10:42:01 tester weewxd[11871]: weewx[11871] INFO user.weatherlink_live: Emitting push (broadcast) packet
Oct 24 10:42:03 tester weewxd[11871]: weewx[11871] INFO user.weatherlink_live: Emitting push (broadcast) packet
Oct 24 10:42:06 tester weewxd[11871]: weewx[11871] INFO user.weatherlink_live: Emitting push (broadcast) packet
Oct 24 10:42:08 tester weewxd[11871]: weewx[11871] INFO user.weatherlink_live: Emitting push (broadcast) packet
Oct 24 10:42:08 tester weewxd[11871]: weewx[11871] INFO user.weatherlink_live: Emitting poll packet
Oct 24 10:42:11 tester weewxd[11871]: weewx[11871] INFO user.weatherlink_live: Emitting push (broadcast) packet
Oct 24 10:42:13 tester weewxd[11871]: weewx[11871] INFO user.weatherlink_live: Emitting push (broadcast) packet
Oct 24 10:42:16 tester weewxd[11871]: weewx[11871] INFO user.weatherlink_live: Emitting push (broadcast) packet
Oct 24 10:42:18 tester weewxd[11871]: weewx[11871] INFO user.weatherlink_live: Emitting push (broadcast) packet
Oct 24 10:42:18 tester weewxd[11871]: weewx[11871] INFO user.weatherlink_live: Emitting poll packet
Oct 24 10:42:21 tester weewxd[11871]: weewx[11871] INFO user.weatherlink_live: Emitting push (broadcast) packet
`

michael-slx, 2020-10-26:
The issue regarding the logging messages should be fixed in the next release.

---

Also thanks for the reminder on the `beaufort` change.

Since the driver doesn't do anything with the `beaufort` type, it seems that the warning message is caused by the configuration option `beaufort = prefer_hardware` in the `[StdWXCalculate]` section. The warning disappears when this line is removed.

I'll removed it from the testing config.

gary-hammer, 2020-10-26:
This was not fixed in 1.0.5
I have this in the stanza:

`[WeatherLinkLive]
    log_success = false`

michael-slx, 2020-10-27:
Ah, ok.

This driver doesn't support per-driver logging options, only the system-wide `log_success` as well as `log_failure` and `debug` options are supported.

I.e. at the top of the `weewx.conf` file:

```ini
# Set to 1 for extra debug info, otherwise comment it out or set to zero
debug = 0

# Whether to log successful operations
log_success = False

# Whether to log unsuccessful operations
log_failure = True
```

As far as I can tell, these options are being phased out anyway with the intention of implementing a [more fine-grained log config system](https://github.com/weewx/weewx/blob/master/docs/logging.md).

closed 2020-10-27 by michael-slx