michael-slx/weewx-weatherlink-live #5: V1.0.5 error message in install

roe-dl, 2020-10-26:
I got an error message when I tried to install the driver:

user@LokalWiki:~$ wee_extension --install weewx-weatherlink-live-v1.0.5.tar.xz 
Request to install 'weewx-weatherlink-live-v1.0.5.tar.xz'
Extracting from tar archive weewx-weatherlink-live-v1.0.5.tar.xz
Traceback (most recent call last):
  File "/usr/share/weewx/wee_extension", line 88, in <module>
    main()
  File "/usr/share/weewx/wee_extension", line 80, in main
    ext.install_extension(options.install)
  File "/usr/share/weewx/weecfg/extension.py", line 123, in install_extension
    self.install_from_dir(extension_dir)
  File "/usr/share/weewx/weecfg/extension.py", line 181, in install_from_dir
    shutil.copy(source_path, destination_path)
  File "/usr/lib/python3.8/shutil.py", line 415, in copy
    copyfile(src, dst, follow_symlinks=follow_symlinks)
  File "/usr/lib/python3.8/shutil.py", line 261, in copyfile
    with open(src, 'rb') as fsrc, open(dst, 'wb') as fdst:
FileNotFoundError: [Errno 2] No such file or directory: '/usr/share/weewx/user/weatherlink_live/__init__.py'

michael-slx, 2020-10-30:
I'm able to reproduce your issue when my user account has insufficient access rights.

Please try to execute `wee_extension` with `sudo` or similar. I.e:

```sh
sudo wee_extension --install weewx-weatherlink-live-v1.0.5.tar.xz /etc/weewx/weewx
.conf
```

roe-dl, 2020-10-31:
Sorry. You are right. I was not careful.

michael-slx, 2020-10-31:
I'm glad I could help you.

closed 2020-10-31 by michael-slx