michael-slx/weewx-weatherlink-live #6: Driver crashes and WeeWx cannot recover

selectSt4r, 2021-03-21:
See log.  I had WeeWx set to capture the log in debug detail.  The first few hundred lines show normal behavior.  The crash stack is toward the end.
[weewx weatherlink-live crashlog.txt](https://github.com/michael-slx/weewx-weatherlink-live/files/6177854/weewx.weatherlink-live.crashlog.txt)

michael-slx, 2021-03-22:
I see you are using the Weather34 skin. Could you try to run WeeWX without the skin (at least with the services disabled), just to see if the issues still appear?

What Pi model are you using?

selectSt4r, 2021-03-23:
Hi Michael -- I'm running it on a Pi 4 with Raspbian 10 Buster.  I will remove Weather34 and report back in a few days.  The crash sometimes takes a few days to happen.  It's happened a few times.

michael-slx, 2021-04-02:
Your log shows that WeeWX could not start any more threads. The driver only starts 2 or so threads and always checks if the old ones have finished before starting new ones. I've checked this by profiling/debugging the driver for 2-3 hours.

So, I've guessed it was caused by Weather34.

closed 2021-04-02 by michael-slx