michael-slx/weewx-weatherlink-live #7: Rain data does not match WLL records

selectSt4r, 2021-03-21:
The rain totals from WLL do not match what is captured by the driver.  The driver, if it doesn't match, is always slightly lower than what WLL reports.

[image]
[image]

selectSt4r, 2021-03-21:
The top screenshot, 11 rows down, is where 11MAR, and corresponds to the WeeWx Day 11.

selectSt4r, 2021-03-30:
After some investigation, I believe there are a couple issues at play.  First, some rainfall is not being archived.  It's usually just of by 0.01" or near that amount.  Next, if it's raining during the start of a new day, the rainfall amounts appear to be inaccurate.  It seems that the driver (or WeeWx, maybe) is unable to discern where (i.e., to which day) to allocate the rain amount archive values.

michael-slx, 2021-04-02:
Can you confirm that this only happens when it rains at midnight? A log with the debug output enabled would be helpful, if possible during midnight when the issue happens.

selectSt4r, 2021-05-22:
Attached is a file showing rain data from 09-MAY-2021.  I don't have logs.  I will have to wait until a good rain is in the forecast and then I will enable some debug-level logging.  I extracted data from WLL and joined it to WeeWX produced data.  If the Rain amount value for either data source was greater than zero, I returned the record.  I have two columns that indicate if there are mismatches between Rain Total and Rain Rate values for a timestamp.

[rainDataMismatches.xlsx](https://github.com/michael-slx/weewx-weatherlink-live/files/6527331/rainDataMismatches.xlsx)

michael-slx, 2021-05-23:
OK, thanks for the spreadsheet.

It seems that for those rows where the data doesn't match, the amount is always `0` and the rate is half the WLL's value. I've experienced a similar issue on my end, but it hasn't occurred since I've enabled the debug log either. It seems there is a crash that causes the driver to skip a packet.

For me, the rain amount field at the time of the crash was `NULL`. Can you check in your db if it's the same on your end?

selectSt4r, 2021-05-23:
Good observation on the ratio for the rain rate.  Didn't notice that.  Are you saying that the raw sensor data that you're retrieving from WLL is written somewhere in the DB?  The most granular table that I could see was the weewx.archive.

michael-slx, 2021-05-23:
No, I'm talking about the `weewx.archive` table. I'm asking because in my case the rain amount was actually `NULL` but got converted to `0` while exporting data to a CSV.

selectSt4r, 2021-05-23:
Just checked and they're `0` in the DB, not `NULL`.

michael-slx, 2021-05-23:
OK, thanks. I'll continue to look into this.

selectSt4r, 2021-06-01:
Michael -- I have a good day of rain ahead of me.  Besides setting the logging to verbose in WeeWX, was there any other logging you need me to enable?

michael-slx, 2021-06-01:
That should be fine. Just set `debug = 1`, `log_success = True` and `log_failure = 1`.

selectSt4r, 2021-06-01:
Logging is set per your request.  FYI, log_failure is boolean like log_success.

michael-slx, 2021-06-06:
Hi @selectSt4r,

I've just published a new [development release](https://github.com/michael-slx/weewx-weatherlink-live/releases/tag/SNAPSHOT-2021-06-06) with some added log messages with regard to the UDP protocol.

Could you please run this version with debug logging on your end and send me a log dump? It doesn't matter whether or not there actually is any rain, I'm just interested in how the driver behaves on your end.

Specifically, I've added a log message showing how many bytes were received from what address.

selectSt4r, 2021-06-07:
Michael -- I've updated the extension to that dev release.  I'm logging with debug = 1.  BTW, the last two rainfalls -- no issues with the data.  Of course not :)

selectSt4r, 2021-06-11:
[selectSt4r_20210611.txt](https://github.com/michael-slx/weewx-weatherlink-live/files/6639727/selectSt4r_20210611.txt)
Michael -- this debug log file was generated using Snapshot 2021-06-06.

selectSt4r, 2021-06-21:
Rain data has been accurate for the last few rainstorms (using Snapshot 2021-06-06).  I will continue to observe now that I've just updated to v1.0.7.

selectSt4r, 2021-08-28:
I did not change anything in my config.  I am currently running v1.0.7 of the extension.  I will try upgrading.  Here is my specific config:

`[WeatherLinkLive]
    driver = user.weatherlink_live
    host = <IP address of host>
    polling_interval = 10
    mapping = th:1, th_indoor, baro, rain:1, wind:1, solar:1, uv:1, thw:1, thsw:1, windchill:1`

LMSSonos, 2021-08-30:
I changed now my config from

`mapping = th:7, th_indoor, baro, rain:7, wind:7, uv:7, solar:7, thw:7, thsw:7:appTemp, windchill:7, battery:7:outTemp:rain:wind:uv`

to 
`mapping = th:7, th_indoor, baro, rain:7, wind:7, uv:7, solar:7, thw:7, thsw:7:appTemp, windchill:7`

now the battery related error has gone, now I only have this one left ;-) I'm using v1.0.10

```
Aug 30 08:22:59 weewx[19] CRITICAL user.rtgd: Unexpected exception of type <class 'TypeError'>
Aug 30 08:22:59 weewx[19] DEBUG user.rtgd: rtgdthread: **** Traceback (most recent call last):
Aug 30 08:22:59 weewx[19] DEBUG user.rtgd: rtgdthread: ****   File "/home/weewx/bin/user/rtgd.py", line 1750, in run
Aug 30 08:22:59 weewx[19] DEBUG user.rtgd: rtgdthread: ****     self.process_packet(_package['payload'])
Aug 30 08:22:59 weewx[19] DEBUG user.rtgd: rtgdthread: ****   File "/home/weewx/bin/user/rtgd.py", line 1805, in process_packet
Aug 30 08:22:59 weewx[19] DEBUG user.rtgd: rtgdthread: ****     cached_packet = self.packet_cache.get_packet(_conv_packet['dateTime'],
Aug 30 08:22:59 weewx[19] DEBUG user.rtgd: rtgdthread: ****   File "/home/weewx/bin/user/rtgd.py", line 2834, in get_packet
Aug 30 08:22:59 weewx[19] DEBUG user.rtgd: rtgdthread: ****     packet[obs] = self.get_value(obs, ts, max_age)
Aug 30 08:22:59 weewx[19] DEBUG user.rtgd: rtgdthread: ****   File "/home/weewx/bin/user/rtgd.py", line 2820, in get_value
Aug 30 08:22:59 weewx[19] DEBUG user.rtgd: rtgdthread: ****     if obs in self.cache and ts - self.cache[obs]['ts'] <= max_age:
Aug 30 08:22:59 weewx[19] DEBUG user.rtgd: rtgdthread: **** TypeError: '<=' not supported between instances of 'int' and 'str'
Aug 30 08:22:59 weewx[19] CRITICAL user.rtgd: Thread exiting. Reason: '<=' not supported between instances of 'int' and 'str'
Aug 30 08:23:03 weewx[19] WARNING user.weatherlink_live: No data since 1 iterations
```

selectSt4r, 2021-08-30:
This looks like an issue:  `thsw:7:appTemp`
Why not set this to `thsw:7`?

michael-slx, 2021-08-30:
Sorry @LMSSonos for the late response.

The [first crash](https://github.com/michael-slx/weewx-weatherlink-live/issues/7#issuecomment-907266675) you posted is related to using the new `battery` mapping option with an old version. Updating to [v1.0.10](https://github.com/michael-slx/weewx-weatherlink-live/releases/tag/v1.0.10) should solve that issue. (Though it seems you've already done that in the meantime.)

The [second crash](https://github.com/michael-slx/weewx-weatherlink-live/issues/7#issuecomment-908067252) seems totally unrelated to this driver and is probably caused by the `rtgd` _(realtime gauge-data)_ extension. Try uninstalling the extension and see if the problem still persists.

LMSSonos, 2021-08-30:
@michael-slx Thanks for the reply

The first one disappeard works also with following config, I assume this is ok with 1.0.10.

`mapping = th:7, th_indoor, baro, rain:7, wind:7, uv:7, solar:7, thw:7, thsw:7:appTemp, windchill:7, battery:7:outTemp:rain:wind:uv` 

Is this valid for the battery? The thing is that in the data `batterStatus1`, `uvBatteryStatus` etc is always 0.

Does not output any error. Sorry for the rtgd issue, this was something else I was testing - sorry for that!

michael-slx, 2021-08-30:
Yes, your config looks fine, assuming your sensor is transmitting on id 7.

Is the battery actually low? `0` means OK, `1` means battery low.

LMSSonos, 2021-08-30:
@michael-slx - thanks for the info, I was expecting a voltage value, that's why I thought 0 is not good ;-) Thanks for clarification.

michael-slx, 2021-08-30:
No problem @LMSSonos
I'll close this issue for now

closed 2021-08-30 by michael-slx