michael-slx/weewx-weatherlink-live #12: Question on Errors

damclarkd, 2021-06-18:
Switched to this driver as a couple of the others for WLL would not allow me to properly shut down Weewx. Data seems to be working fine with the exception of these errors in the log. These will show up every 15-20 minutes it seems. I found similar issues of this kind when searching the weewx Goggle Groups but no resolution. 
Wondering if this is normal behavior and the problem might actually be with the network? Or of you might have suggestions?

Thank you.

Jun 18 07:34:51 raspberrypi weewx[3110] ERROR user.weatherlink_live.scheduler: Error caught in scheduler tick. Not rescheduling
Jun 18 07:34:51 raspberrypi weewx[3110] INFO weewx.engine: Main loop exiting. Shutting engine down.
Jun 18 07:34:51 raspberrypi weewx[3110] INFO weewx.engine: Shutting down StdReport thread
Jun 18 07:34:51 raspberrypi weewx[3110] CRITICAL __main__: Caught WeeWxIOError: Error while receiving or processing packets: HTTPConnectionPool(host='10.1.20.40', port=80): Max retries exceeded with url: /v1/real_time?duration=1500 (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0xb22f2770>: Failed to establish a new connection: [Errno 111] Connection refused'))
Jun 18 07:34:51 raspberrypi weewx[3110] CRITICAL __main__:     ****  Waiting 60 seconds then retrying...

michael-slx, 2021-06-18:
Hi,
no, this isn't normal operating behavior.

Do you have any other device or another program talking to the WLL? The WLL is quite easily overwhelmed when too many connections are happening. Maybe a second instance of WeeWX that didn't get shut down property and is hanging around in the background?

Otherwise, the only other idea I have is that it might be something network related, as you suggested.

damclarkd, 2021-06-18:
Michael,
Thank you for your response. As I did not have these errors while trying other drivers for WLL, I thought I would try you first. You bring up some good points.
No, no other programs are polling the WLL. However, I am sending data OUT from the WLL to CWOP every 15 minutes. I am now in the process of moving that over to Weewx and eliminate that task from the WLL. Hopefully that might eliminate my errors.

Cheers,
Dennis

closed 2021-08-20 by damclarkd