michael-slx/weewx-weatherlink-live #14: Changes to support Vantage Vue

thedoctors-cj, 2021-09-23:
I was super happy to find this project to start working with the data from my Vantage Vue.  I tried the default mapping and was getting a "Not a broadcast packet" error for both my RainMapping and my WindMapping in the DEBUG logs.  If I remove

```
        if packet.data_source != PacketSource.WEATHER_PUSH:
            self._log_mapping_notResponsible("Not a broadcast packet")
            return
```
from lines 199-201  and 238-241 in the mappers.py file, I am then able to successfully parse my rain and wind data.  These must be there for supporting some devices.  Is there some way we could integrate a change like this to support Vantage Vue for others to use?

Thank you,
Craig

michael-slx, 2021-09-24:
The log message you're seeing is absolutely normal and not an error. These lines are there to ignore the wind and rain data from the "complete" packets, because these values get sent every 2.5 seconds anyway. 

If you put those removed lines back in, are you getting rain and wind values?

thedoctors-cj, 2021-09-25:
Sorry, I was not clear in my first note.  I do not get any rain or wind values from my WeatherLink unless I comment out those lines.

michael-slx, 2021-09-25:
Ah, ok. That sounds like the driver isn't receiving the 2.5 seconds "live" updates.

How is your local network set up? Are the WeatherLink and the WeeWX device on the same LAN? The network has to allow broadcast packets to pass between the two devices.

nickbp, 2022-03-01:
FWIW I've been using a Vantage Vue successfully with 4.5.1. In my case it took me a while to figure out the UDP broadcast packets, but things finally worked once I had ensured that WeeWX was reachable on the same subnet as the WeatherLink Live. I wrote up some troubleshooting info via this PR: https://github.com/michael-slx/weewx-weatherlink-live/pull/21