michael-slx/weewx-weatherlink-live #15: No reports after installation

groetg, 2021-12-02:
I am using WeatherLink Live at weewx and want to get Soilmoist (channel 2) and Soiltemp in the weewx reports, I use the weewx-weatherlink-live station with the proper mapping (example on github), want there are no reports coming. Log file:

Dec  2 15:33:54 raspberrypi weewx[1948] INFO __main__: retrying...
Dec  2 15:33:54 raspberrypi weewx[1948] INFO __main__: Using configuration file /etc/weewx/weewx.conf
Dec  2 15:33:54 raspberrypi weewx[1948] INFO __main__: Debug is 1
Dec  2 15:33:54 raspberrypi weewx[1948] DEBUG __main__: Initializing engine
Dec  2 15:33:54 raspberrypi weewx[1948] INFO weewx.engine: Loading station type WeatherLinkLive (user.weatherlink_live)
Dec  2 15:33:54 raspberrypi weewx[1948] INFO user.weatherlink_live: Initializing driver: WeatherLinkLive v1.0.10
Dec  2 15:33:54 raspberrypi weewx[1948] DEBUG user.weatherlink_live: Configuration: {'host': '192.168.1.29', 'mappings': [['th', '1'], ['th_indoor'], ['baro'], ['rain', '1'], ['wind', '1'], ['uv', '1'], ['solar', '1'], ['thw', '1'], ['thsw', '1', 'appTemp'], ['windchill', '1'], ['soil_temp', '2', '1'], ['soil_temp', '2', '2'], ['soil_temp', '2', '3'], ['soil_temp', '2', '4'], ['soil_moist', '2', '1'], ['soil_moist', '2', '2'], ['soil_moist', '2', '3'], ['soil_moist', '2', '4'], ['leaf_wet', '2', '1'], ['lef_wet', '2', '2'], ['battery', '1', 'outTemp', 'wind'], ['battery', '2', 'tx']], 'polling_interval': 10.0, 'max_no_data_iterations': 5, 'log_success': True, 'log_error': True, 'socket_timeout': 20.0}
Dec  2 15:33:54 raspberrypi weewx[1948] DEBUG user.weatherlink_live.configuration: Creating mapper th. Options: ['1']
Dec  2 15:33:54 raspberrypi weewx[1948] DEBUG user.weatherlink_live.mappers: THMapping[['1']]: Mapping targets: {'t': 'outTemp', 'h': 'outHumidity', 'dp': 'dewpoint', 'hi': 'heatindex', 'wb': 'wetbulb'}
Dec  2 15:33:54 raspberrypi weewx[1948] DEBUG user.weatherlink_live.configuration: Creating mapper th_indoor. Options: []
Dec  2 15:33:54 raspberrypi weewx[1948] DEBUG user.weatherlink_live.mappers: THIndoorMapping[[]]: Mapping targets: {'t': 'inTemp', 'h': 'inHumidity', 'dp': 'inDewpoint', 'hi': 'inHeatindex'}
Dec  2 15:33:54 raspberrypi weewx[1948] DEBUG user.weatherlink_live.configuration: Creating mapper baro. Options: []
Dec  2 15:33:54 raspberrypi weewx[1948] DEBUG user.weatherlink_live.mappers: BaroMapping[[]]: Mapping targets: {'baro_abs': 'pressure', 'baro_sl': 'altimeter'}
Dec  2 15:33:54 raspberrypi weewx[1948] DEBUG user.weatherlink_live.configuration: Creating mapper rain. Options: ['1']
Dec  2 15:33:54 raspberrypi weewx[1948] DEBUG user.weatherlink_live.mappers: RainMapping[['1']]: Mapping targets: {'amount': 'rain', 'rate': 'rainRate', 'count': 'rainCount', 'count_rate': 'rainCountRate', 'size': 'rainSize'}
Dec  2 15:33:54 raspberrypi weewx[1948] DEBUG user.weatherlink_live.configuration: Creating mapper wind. Options: ['1']
Dec  2 15:33:54 raspberrypi weewx[1948] DEBUG user.weatherlink_live.mappers: WindMapping[['1']]: Mapping targets: {'wind_dir': 'windDir', 'wind_speed': 'windSpeed', 'gust_dir': 'windGustDir', 'gust_speed': 'windGust'}
Dec  2 15:33:54 raspberrypi weewx[1948] DEBUG user.weatherlink_live.configuration: Creating mapper uv. Options: ['1']
Dec  2 15:33:54 raspberrypi weewx[1948] DEBUG user.weatherlink_live.mappers: UvMapping[['1']]: Mapping targets: {'uv': 'UV'}
Dec  2 15:33:54 raspberrypi weewx[1948] DEBUG user.weatherlink_live.configuration: Creating mapper solar. Options: ['1']
Dec  2 15:33:54 raspberrypi weewx[1948] DEBUG user.weatherlink_live.mappers: SolarMapping[['1']]: Mapping targets: {'solar': 'radiation'}
Dec  2 15:33:54 raspberrypi weewx[1948] DEBUG user.weatherlink_live.configuration: Creating mapper thw. Options: ['1']
Dec  2 15:33:54 raspberrypi weewx[1948] DEBUG user.weatherlink_live.mappers: ThwMapping[['1']]: Mapping targets: {'thw': 'thw'}
Dec  2 15:33:54 raspberrypi weewx[1948] DEBUG user.weatherlink_live.configuration: Creating mapper thsw. Options: ['1', 'appTemp']
Dec  2 15:33:54 raspberrypi weewx[1948] DEBUG user.weatherlink_live.mappers: ThswMapping[['1', 'appTemp']]: Mapping targets: {'thsw': 'thsw', 'app_temp': 'appTemp'}
Dec  2 15:33:54 raspberrypi weewx[1948] DEBUG user.weatherlink_live.configuration: Creating mapper windchill. Options: ['1']
Dec  2 15:33:54 raspberrypi weewx[1948] DEBUG user.weatherlink_live.mappers: WindChillMapping[['1']]: Mapping targets: {'windchill': 'windchill'}
Dec  2 15:33:54 raspberrypi weewx[1948] DEBUG user.weatherlink_live.configuration: Creating mapper soil_temp. Options: ['2', '1']
Dec  2 15:33:54 raspberrypi weewx[1948] DEBUG user.weatherlink_live.mappers: SoilTempMapping[['2', '1']]: Mapping targets: {'soil_temp': 'soilTemp1'}
Dec  2 15:33:54 raspberrypi weewx[1948] DEBUG user.weatherlink_live.configuration: Creating mapper soil_temp. Options: ['2', '2']
Dec  2 15:33:54 raspberrypi weewx[1948] DEBUG user.weatherlink_live.mappers: SoilTempMapping[['2', '2']]: Mapping targets: {'soil_temp': 'soilTemp2'}
Dec  2 15:33:54 raspberrypi weewx[1948] DEBUG user.weatherlink_live.configuration: Creating mapper soil_temp. Options: ['2', '3']
Dec  2 15:33:54 raspberrypi weewx[1948] DEBUG user.weatherlink_live.mappers: SoilTempMapping[['2', '3']]: Mapping targets: {'soil_temp': 'soilTemp3'}
Dec  2 15:33:54 raspberrypi weewx[1948] DEBUG user.weatherlink_live.configuration: Creating mapper soil_temp. Options: ['2', '4']
Dec  2 15:33:54 raspberrypi weewx[1948] DEBUG user.weatherlink_live.mappers: SoilTempMapping[['2', '4']]: Mapping targets: {'soil_temp': 'soilTemp4'}
Dec  2 15:33:54 raspberrypi weewx[1948] DEBUG user.weatherlink_live.configuration: Creating mapper soil_moist. Options: ['2', '1']
Dec  2 15:33:54 raspberrypi weewx[1948] DEBUG user.weatherlink_live.mappers: SoilMoistureMapping[['2', '1']]: Mapping targets: {'soil_moisture': 'soilMoist1'}
Dec  2 15:33:54 raspberrypi weewx[1948] DEBUG user.weatherlink_live.configuration: Creating mapper soil_moist. Options: ['2', '2']
Dec  2 15:33:54 raspberrypi weewx[1948] DEBUG user.weatherlink_live.mappers: SoilMoistureMapping[['2', '2']]: Mapping targets: {'soil_moisture': 'soilMoist2'}
Dec  2 15:33:54 raspberrypi weewx[1948] DEBUG user.weatherlink_live.configuration: Creating mapper soil_moist. Options: ['2', '3']
Dec  2 15:33:54 raspberrypi weewx[1948] DEBUG user.weatherlink_live.mappers: SoilMoistureMapping[['2', '3']]: Mapping targets: {'soil_moisture': 'soilMoist3'}
Dec  2 15:33:54 raspberrypi weewx[1948] DEBUG user.weatherlink_live.configuration: Creating mapper soil_moist. Options: ['2', '4']
Dec  2 15:33:54 raspberrypi weewx[1948] DEBUG user.weatherlink_live.mappers: SoilMoistureMapping[['2', '4']]: Mapping targets: {'soil_moisture': 'soilMoist4'}
Dec  2 15:33:54 raspberrypi weewx[1948] DEBUG user.weatherlink_live.configuration: Creating mapper leaf_wet. Options: ['2', '1']
Dec  2 15:33:54 raspberrypi weewx[1948] ERROR weewx.engine: Import of driver failed: 'set' object is not subscriptable (<class 'TypeError'>)
Dec  2 15:33:54 raspberrypi weewx[1948] CRITICAL weewx.engine:     ****  Traceback (most recent call last):
Dec  2 15:33:54 raspberrypi weewx[1948] CRITICAL weewx.engine:     ****    File "/usr/share/weewx/weewx/engine.py", line 119, in setupStation
Dec  2 15:33:54 raspberrypi weewx[1948] CRITICAL weewx.engine:     ****      self.console = loader_function(config_dict, self)
Dec  2 15:33:54 raspberrypi weewx[1948] CRITICAL weewx.engine:     ****    File "/usr/share/weewx/user/weatherlink_live/__init__.py", line 100, in loader
Dec  2 15:33:54 raspberrypi weewx[1948] CRITICAL weewx.engine:     ****      return WeatherlinkLiveDriver(config_dict, engine)
Dec  2 15:33:54 raspberrypi weewx[1948] CRITICAL weewx.engine:     ****    File "/usr/share/weewx/user/weatherlink_live/__init__.py", line 121, in __init__
Dec  2 15:33:54 raspberrypi weewx[1948] CRITICAL weewx.engine:     ****      self.mappers = self.configuration.create_mappers()
Dec  2 15:33:54 raspberrypi weewx[1948] CRITICAL weewx.engine:     ****    File "/usr/share/weewx/user/weatherlink_live/configuration.py", line 116, in create_mappers
Dec  2 15:33:54 raspberrypi weewx[1948] CRITICAL weewx.engine:     ****      mapper = self._create_mapper(source_opts, used_record_keys)
Dec  2 15:33:54 raspberrypi weewx[1948] CRITICAL weewx.engine:     ****    File "/usr/share/weewx/user/weatherlink_live/configuration.py", line 132, in _create_mapper
[... cut ...]
Dec  2 15:33:54 raspberrypi weewx[1948] CRITICAL weewx.engine:     ****    File "/usr/share/weewx/user/weatherlink_live/mappers.py", line 97, in <listcomp>
Dec  2 15:33:54 raspberrypi weewx[1948] CRITICAL weewx.engine:     ****      (k, v[i]) for k, v in available_map_targets_dict.items()
Dec  2 15:33:54 raspberrypi weewx[1948] CRITICAL weewx.engine:     ****  TypeError: 'set' object is not subscriptable
Dec  2 15:33:54 raspberrypi weewx[1948] CRITICAL __main__: Unable to load driver: 'set' object is not subscriptable
Dec  2 15:33:54 raspberrypi weewx[1948] CRITICAL __main__:     ****  Waiting 60 seconds then retrying...

michael-slx, 2021-12-03:
Should be fixed in release v1.0.11

groetg, 2021-12-03:
After installing v1.0.11, same syslog error:

Dec  3 21:02:26 raspberrypi weewx[630] INFO __main__: Using configuration file /etc/weewx/weewx.conf
Dec  3 21:02:26 raspberrypi weewx[630] INFO __main__: Debug is 1
Dec  3 21:02:26 raspberrypi weewx[630] DEBUG __main__: Initializing engine
Dec  3 21:02:26 raspberrypi weewx[630] INFO weewx.engine: Loading station type WeatherLinkLive (user.weatherlink_live)
Dec  3 21:02:26 raspberrypi weewx[630] INFO user.weatherlink_live: Initializing driver: WeatherLinkLive v1.0.10
Dec  3 21:02:26 raspberrypi weewx[630] DEBUG user.weatherlink_live: Configuration: {'host': '192.168.1.29', 'mappings': [['th', '1'], ['th_indoor'], ['baro'], ['rain', '1'], ['wind', '1'], ['uv', '1'], ['solar', '1'], ['thw', '1'], ['thsw', '1', 'appTemp'], ['windchill', '1'], ['soil_temp', '2', '1'], ['soil_temp', '2', '2'], ['soil_temp', '2', '3'], ['soil_temp', '2', '4'], ['soil_moist', '2', '1'], ['soil_moist', '2', '2'], ['soil_moist', '2', '3'], ['soil_moist', '2', '4'], ['leaf_wet', '2', '1'], ['lef_wet', '2', '2'], ['battery', '1', 'outTemp', 'wind'], ['battery', '2', 'tx']], 'polling_interval': 10.0, 'max_no_data_iterations': 5, 'log_success': True, 'log_error': True, 'socket_timeout': 20.0}
Dec  3 21:02:26 raspberrypi weewx[630] DEBUG user.weatherlink_live.configuration: Creating mapper th. Options: ['1']
Dec  3 21:02:26 raspberrypi weewx[630] DEBUG user.weatherlink_live.mappers: THMapping[['1']]: Mapping targets: {'t': 'outTemp', 'h': 'outHumidity', 'dp': 'dewpoint', 'hi': 'heatindex', 'wb': 'wetbulb'}
Dec  3 21:02:26 raspberrypi weewx[630] DEBUG user.weatherlink_live.configuration: Creating mapper th_indoor. Options: []
Dec  3 21:02:26 raspberrypi weewx[630] DEBUG user.weatherlink_live.mappers: THIndoorMapping[[]]: Mapping targets: {'t': 'inTemp', 'h': 'inHumidity', 'dp': 'inDewpoint', 'hi': 'inHeatindex'}
Dec  3 21:02:26 raspberrypi weewx[630] DEBUG user.weatherlink_live.configuration: Creating mapper baro. Options: []
Dec  3 21:02:26 raspberrypi weewx[630] DEBUG user.weatherlink_live.mappers: BaroMapping[[]]: Mapping targets: {'baro_abs': 'pressure', 'baro_sl': 'altimeter'}
Dec  3 21:02:26 raspberrypi weewx[630] DEBUG user.weatherlink_live.configuration: Creating mapper rain. Options: ['1']
Dec  3 21:02:26 raspberrypi weewx[630] DEBUG user.weatherlink_live.mappers: RainMapping[['1']]: Mapping targets: {'amount': 'rain', 'rate': 'rainRate', 'count': 'rainCount', 'count_rate': 'rainCountRate', 'size': 'rainSize'}
Dec  3 21:02:26 raspberrypi weewx[630] DEBUG user.weatherlink_live.configuration: Creating mapper wind. Options: ['1']
Dec  3 21:02:26 raspberrypi weewx[630] DEBUG user.weatherlink_live.mappers: WindMapping[['1']]: Mapping targets: {'wind_dir': 'windDir', 'wind_speed': 'windSpeed', 'gust_dir': 'windGustDir', 'gust_speed': 'windGust'}
Dec  3 21:02:26 raspberrypi weewx[630] DEBUG user.weatherlink_live.configuration: Creating mapper uv. Options: ['1']
Dec  3 21:02:26 raspberrypi weewx[630] DEBUG user.weatherlink_live.mappers: UvMapping[['1']]: Mapping targets: {'uv': 'UV'}
Dec  3 21:02:26 raspberrypi weewx[630] DEBUG user.weatherlink_live.configuration: Creating mapper solar. Options: ['1']
Dec  3 21:02:26 raspberrypi weewx[630] DEBUG user.weatherlink_live.mappers: SolarMapping[['1']]: Mapping targets: {'solar': 'radiation'}
Dec  3 21:02:26 raspberrypi weewx[630] DEBUG user.weatherlink_live.configuration: Creating mapper thw. Options: ['1']
Dec  3 21:02:26 raspberrypi weewx[630] DEBUG user.weatherlink_live.mappers: ThwMapping[['1']]: Mapping targets: {'thw': 'thw'}
Dec  3 21:02:26 raspberrypi weewx[630] DEBUG user.weatherlink_live.configuration: Creating mapper thsw. Options: ['1', 'appTemp']
Dec  3 21:02:26 raspberrypi weewx[630] DEBUG user.weatherlink_live.mappers: ThswMapping[['1', 'appTemp']]: Mapping targets: {'thsw': 'thsw', 'app_temp': 'appTemp'}
Dec  3 21:02:26 raspberrypi weewx[630] DEBUG user.weatherlink_live.configuration: Creating mapper windchill. Options: ['1']
Dec  3 21:02:26 raspberrypi weewx[630] DEBUG user.weatherlink_live.mappers: WindChillMapping[['1']]: Mapping targets: {'windchill': 'windchill'}
Dec  3 21:02:26 raspberrypi weewx[630] DEBUG user.weatherlink_live.configuration: Creating mapper soil_temp. Options: ['2', '1']
Dec  3 21:02:26 raspberrypi weewx[630] DEBUG user.weatherlink_live.mappers: SoilTempMapping[['2', '1']]: Mapping targets: {'soil_temp': 'soilTemp1'}
Dec  3 21:02:26 raspberrypi weewx[630] DEBUG user.weatherlink_live.configuration: Creating mapper soil_temp. Options: ['2', '2']
Dec  3 21:02:26 raspberrypi weewx[630] DEBUG user.weatherlink_live.mappers: SoilTempMapping[['2', '2']]: Mapping targets: {'soil_temp': 'soilTemp2'}
Dec  3 21:02:26 raspberrypi weewx[630] DEBUG user.weatherlink_live.configuration: Creating mapper soil_temp. Options: ['2', '3']
Dec  3 21:02:26 raspberrypi weewx[630] DEBUG user.weatherlink_live.mappers: SoilTempMapping[['2', '3']]: Mapping targets: {'soil_temp': 'soilTemp3'}
Dec  3 21:02:26 raspberrypi weewx[630] DEBUG user.weatherlink_live.configuration: Creating mapper soil_temp. Options: ['2', '4']
Dec  3 21:02:26 raspberrypi weewx[630] DEBUG user.weatherlink_live.mappers: SoilTempMapping[['2', '4']]: Mapping targets: {'soil_temp': 'soilTemp4'}
Dec  3 21:02:26 raspberrypi weewx[630] DEBUG user.weatherlink_live.configuration: Creating mapper soil_moist. Options: ['2', '1']
[... cut ...]
Dec  3 21:02:26 raspberrypi weewx[630] CRITICAL weewx.engine:     ****    File "/usr/share/weewx/user/weatherlink_live/mappers.py", line 97, in <listcomp>
Dec  3 21:02:26 raspberrypi weewx[630] CRITICAL weewx.engine:     ****      (k, v[i]) for k, v in available_map_targets_dict.items()
Dec  3 21:02:26 raspberrypi weewx[630] CRITICAL weewx.engine:     ****  TypeError: 'set' object is not subscriptable
Dec  3 21:02:26 raspberrypi weewx[630] CRITICAL __main__: Unable to load driver: 'set' object is not subscriptable
Dec  3 21:02:26 raspberrypi weewx[630] CRITICAL __main__:     ****  Waiting 60 seconds then retrying...

michael-slx, 2021-12-04:
The logs show driver version 1.9.10 is used, i.e. the old one.

```
 Dec 3 21:02:26 raspberrypi weewx[630] INFO user.weatherlink_live: Initializing driver: WeatherLinkLive v1.0.10
```

Could you try to uninstall the driver using the [`wee_extension`](https://weewx.com/docs/utilities.htm#wee_extension_utility) tool and then reinstall the new version?

groetg, 2021-12-04:
the lef_wet error tayed, so I searched where the lef_wet typo came from. It origin was in de mapping line (I copied from this site) in weewx.conf file. Probably made the typo myself i guess?

After restart, the driver works, but gets still no soilmoist data:

log:
Dec  4 15:13:06 raspberrypi weewx[2288] DEBUG user.weatherlink_live.davis_broadcast: Received 408 bytes from ('192.168.1.29', 51574)
Dec  4 15:13:06 raspberrypi weewx[2288] DEBUG user.weatherlink_live.packets: Trying to create UDP conditions packet
Dec  4 15:13:06 raspberrypi weewx[2288] DEBUG user.weatherlink_live.data_host: Received new broadcast packet
Dec  4 15:13:06 raspberrypi weewx[2288] DEBUG user.weatherlink_live.mappers: THMapping[['1']]: Observation not found in packet
Dec  4 15:13:06 raspberrypi weewx[2288] DEBUG user.weatherlink_live.mappers: THIndoorMapping[[]]: Observation not found in packet
Dec  4 15:13:06 raspberrypi weewx[2288] DEBUG user.weatherlink_live.mappers: BaroMapping[[]]: Observation not found in packet
Dec  4 15:13:06 raspberrypi weewx[2288] DEBUG user.weatherlink_live.mappers: RainMapping[['1']]: Mapped: rainSize=0.007874015748031496
Dec  4 15:13:06 raspberrypi weewx[2288] DEBUG user.weatherlink_live.mappers: RainMapping[['1']]: Mapped: rainCountRate=5
Dec  4 15:13:06 raspberrypi weewx[2288] DEBUG user.weatherlink_live.mappers: RainMapping[['1']]: Mapped: rainRate=0.03937007874015748
Dec  4 15:13:06 raspberrypi weewx[2288] DEBUG user.weatherlink_live.mappers: RainMapping[['1']]: Mapped: rainCount=0
Dec  4 15:13:06 raspberrypi weewx[2288] DEBUG user.weatherlink_live.mappers: RainMapping[['1']]: Mapped: rain=0.0
Dec  4 15:13:06 raspberrypi weewx[2288] DEBUG user.weatherlink_live.mappers: WindMapping[['1']]: Mapped: windDir=0
Dec  4 15:13:06 raspberrypi weewx[2288] DEBUG user.weatherlink_live.mappers: WindMapping[['1']]: Mapped: windSpeed=0.0
Dec  4 15:13:06 raspberrypi weewx[2288] DEBUG user.weatherlink_live.mappers: UvMapping[['1']]: Observation not found in packet
Dec  4 15:13:06 raspberrypi weewx[2288] DEBUG user.weatherlink_live.mappers: SolarMapping[['1']]: Observation not found in packet
Dec  4 15:13:06 raspberrypi weewx[2288] DEBUG user.weatherlink_live.mappers: ThwMapping[['1']]: Observation not found in packet
Dec  4 15:13:06 raspberrypi weewx[2288] DEBUG user.weatherlink_live.mappers: ThswMapping[['1', 'appTemp']]: Observation not found in packet
Dec  4 15:13:06 raspberrypi weewx[2288] DEBUG user.weatherlink_live.mappers: WindChillMapping[['1']]: Observation not found in packet
Dec  4 15:13:06 raspberrypi weewx[2288] DEBUG user.weatherlink_live.mappers: SoilTempMapping[['1', '1']]: Observation not found in packet
Dec  4 15:13:06 raspberrypi weewx[2288] DEBUG user.weatherlink_live.mappers: SoilMoistureMapping[['1', '1']]: Observation not found in packet
Dec  4 15:13:06 raspberrypi weewx[2288] DEBUG user.weatherlink_live.mappers: BatteryStatusMapping[['1', 'outTemp', 'wind']]: Observation not found in packet
Dec  4 15:13:06 raspberrypi weewx[2288] DEBUG user.weatherlink_live.mappers: BatteryStatusMapping[['2', 'tx']]: Observation not found in packet
Dec  4 15:13:06 raspberrypi weewx[2288] DEBUG user.weatherlink_live: Emitting push (broadcast) packet
Dec  4 15:13:06 raspberrypi weewx[2288] DEBUG user.weatherlink_live.service: WllWindGustService: Updating record with dict: {'windGust': 4.0, 'windGustDir': 204}
Dec  4 15:13:06 raspberrypi weewx[2288] DEBUG user.weatherlink_live: Waiting for new packet

mapping:
mapping = th:1, th_indoor, baro, rain:1, wind:1, uv:1, solar:1, thw:1, thsw:1:appTemp, windchill:1, soil_temp:1:1, soil_moist:1:1, battery:1:outTemp:wind, battery:2:tx

Soilmoist is on channel 2 in Davis Vantage (ISS is channel 1)

When i looked at the API url from my weatherlink live, the iss seems to be at #0 and the soilmoist at #1:
 	
data	
did	"@@"
ts	1638625679
conditions	
0	
lsid	470641
data_structure_type	1
txid	1
temp	41.5
hum	96.4
dew_point	40.6
wet_bulb	41
heat_index	41.5
wind_chill	41.5
thw_index	41.5
thsw_index	39.5
wind_speed_last	2
wind_dir_last	209
wind_speed_avg_last_1_min	1.62
wind_dir_scalar_avg_last_1_min	209
wind_speed_avg_last_2_min	1.12
wind_dir_scalar_avg_last_2_min	209
wind_speed_hi_last_2_min	4
wind_dir_at_hi_speed_last_2_min	132
wind_speed_avg_last_10_min	0.62
wind_dir_scalar_avg_last_10_min	225
wind_speed_hi_last_10_min	5
wind_dir_at_hi_speed_last_10_min	216
rain_size	2
rain_rate_last	1
rain_rate_hi	1
rainfall_last_15_min	1
rain_rate_hi_last_15_min	1
rainfall_last_60_min	2
rainfall_last_24_hr	38
rain_storm	374
rain_storm_start_at	1637821740
solar_rad	12
uv_index	0
rx_state	0
trans_battery_flag	0
rainfall_daily	2
rainfall_monthly	121
rainfall_year	385
rain_storm_last	11
rain_storm_last_start_at	1637451960
rain_storm_last_end_at	1637586061
1	
lsid	470642
data_structure_type	2
txid	2
**temp_1	44.3**
temp_2	null
temp_3	null
temp_4	null
**moist_soil_1	26.2**
moist_soil_2	null
moist_soil_3	null
moist_soil_4	null
wet_leaf_1	null
wet_leaf_2	null
rx_state	0
trans_battery_flag	0
2	
lsid	470639
data_structure_type	4
temp_in	76.5
hum_in	39.4
dew_point_in	50
heat_index_in	75.4
3	
lsid	470638
data_structure_type	3
bar_sea_level	29.321
bar_trend	-0.044
bar_absolute	29.32
error	null

I am a little bit a noob i guess??

My goal is to get Soilmoist in my neo-material skin and in charts....

michael-slx, 2021-12-04:
The typo in the mapping line was my mistake, I've already fixed it yesterday.

The mapping for your setup is:

```ini
 mapping = th:1, th_indoor, baro, rain:1, wind:1, uv:1, solar:1, thw:1, thsw:1:appTemp, windchill:1, soil_temp:2:1, soil_moist:2:1, battery:1:outTemp:wind, battery:2:tx
```

Note that I've changed the `1` to a `2` after `soil_temp` and `soil_moist`.

groetg, 2021-12-04:
It works!! Thanx. The values soiloist are mapped now. The following problem is to get it visible in the neoWX material skin.

groetg, 2021-12-04:
Dec  4 15:43:17 raspberrypi weewx[2752] DEBUG user.weatherlink_live.scheduler: Notifying poll callback
Dec  4 15:43:17 raspberrypi weewx[2752] DEBUG urllib3.connectionpool: Starting new HTTP connection (1): 192.168.1.29:80
Dec  4 15:43:17 raspberrypi weewx[2752] DEBUG urllib3.connectionpool: http://192.168.1.29:80 "GET /v1/current_conditions HTTP/1.1" 200 None
Dec  4 15:43:17 raspberrypi weewx[2752] DEBUG user.weatherlink_live.packets: Trying to create HTTP conditions packet
Dec  4 15:43:17 raspberrypi weewx[2752] DEBUG user.weatherlink_live.data_host: Polled current conditions
Dec  4 15:43:17 raspberrypi weewx[2752] DEBUG user.weatherlink_live.mappers: THMapping[['1']]: Mapped: outTemp=42.3
Dec  4 15:43:17 raspberrypi weewx[2752] DEBUG user.weatherlink_live.mappers: THMapping[['1']]: Mapped: outHumidity=93.5
Dec  4 15:43:17 raspberrypi weewx[2752] DEBUG user.weatherlink_live.mappers: THMapping[['1']]: Mapped: dewpoint=40.6
Dec  4 15:43:17 raspberrypi weewx[2752] DEBUG user.weatherlink_live.mappers: THMapping[['1']]: Mapped: heatindex=42.3
Dec  4 15:43:17 raspberrypi weewx[2752] DEBUG user.weatherlink_live.mappers: THMapping[['1']]: Mapped: wetbulb=41.4
Dec  4 15:43:17 raspberrypi weewx[2752] DEBUG user.weatherlink_live.mappers: THIndoorMapping[[]]: Mapped: inTemp=76.8
Dec  4 15:43:17 raspberrypi weewx[2752] DEBUG user.weatherlink_live.mappers: THIndoorMapping[[]]: Mapped: inHumidity=39.0
Dec  4 15:43:17 raspberrypi weewx[2752] DEBUG user.weatherlink_live.mappers: THIndoorMapping[[]]: Mapped: inDewpoint=50.0
Dec  4 15:43:17 raspberrypi weewx[2752] DEBUG user.weatherlink_live.mappers: THIndoorMapping[[]]: Mapped: inHeatindex=75.7
Dec  4 15:43:17 raspberrypi weewx[2752] DEBUG user.weatherlink_live.mappers: BaroMapping[[]]: Mapped: pressure=29.324
Dec  4 15:43:17 raspberrypi weewx[2752] DEBUG user.weatherlink_live.mappers: BaroMapping[[]]: Mapped: altimeter=29.325
Dec  4 15:43:17 raspberrypi weewx[2752] DEBUG user.weatherlink_live.mappers: RainMapping[['1']]: Mapping not responsible: Not a broadcast packet
Dec  4 15:43:17 raspberrypi weewx[2752] DEBUG user.weatherlink_live.mappers: WindMapping[['1']]: Mapping not responsible: Not a broadcast packet
Dec  4 15:43:17 raspberrypi weewx[2752] DEBUG user.weatherlink_live.mappers: UvMapping[['1']]: Mapped: UV=0.0
Dec  4 15:43:17 raspberrypi weewx[2752] DEBUG user.weatherlink_live.mappers: SolarMapping[['1']]: Mapped: radiation=0
Dec  4 15:43:17 raspberrypi weewx[2752] DEBUG user.weatherlink_live.mappers: ThwMapping[['1']]: Mapped: thw=42.3
Dec  4 15:43:17 raspberrypi weewx[2752] DEBUG user.weatherlink_live.mappers: ThswMapping[['1', 'appTemp']]: Mapped: thsw=40.3
Dec  4 15:43:17 raspberrypi weewx[2752] DEBUG user.weatherlink_live.mappers: ThswMapping[['1', 'appTemp']]: Mapped: appTemp=40.3
Dec  4 15:43:17 raspberrypi weewx[2752] DEBUG user.weatherlink_live.mappers: WindChillMapping[['1']]: Mapped: windchill=42.3
Dec  4 15:43:17 raspberrypi weewx[2752] DEBUG user.weatherlink_live.mappers: SoilTempMapping[['2', '1']]: Mapped: soilTemp1=44.3
Dec  4 15:43:17 raspberrypi weewx[2752] DEBUG user.weatherlink_live.mappers: SoilMoistureMapping[['2', '1']]: Mapped: soilMoist1=26.2
Dec  4 15:43:17 raspberrypi weewx[2752] DEBUG user.weatherlink_live.mappers: BatteryStatusMapping[['1', 'outTemp', 'wind']]: Mapped: batteryStatus1=0
Dec  4 15:43:17 raspberrypi weewx[2752] DEBUG user.weatherlink_live.mappers: BatteryStatusMapping[['1', 'outTemp', 'wind']]: Mapped: outTempBatteryStatus=0
Dec  4 15:43:17 raspberrypi weewx[2752] DEBUG user.weatherlink_live.mappers: BatteryStatusMapping[['1', 'outTemp', 'wind']]: Mapped: windBatteryStatus=0
Dec  4 15:43:17 raspberrypi weewx[2752] DEBUG user.weatherlink_live.mappers: BatteryStatusMapping[['2', 'tx']]: Mapped: batteryStatus2=0
Dec  4 15:43:17 raspberrypi weewx[2752] DEBUG user.weatherlink_live.mappers: BatteryStatusMapping[['2', 'tx']]: Mapped: txBatteryStatus=0

michael-slx, 2021-12-04:
The log looks OK now.

I don't know anything about the neoWX skin, so I'm afraid I can't really help you.

groetg, 2021-12-04:
The try and error method has worked! I got it running.
[http://www.buroscope.nl/weerstation](url)

Thanx for helping me out

michael-slx, 2021-12-04:
Looks good!

Closing #15

commit 9616bd9c (2021-12-03): Fix map targets specified as sets

Fixes #15
closed 2021-12-04 by michael-slx