michael-slx/weewx-weatherlink-live #16: Question?

pavll, 2021-12-15:
What do you mean with: "Not support historical data support". I thought the data will be saved in weewx db from when it get's connected to weatherstation via WeatherLink Live Access point's local api?

pavll, 2021-12-15:
And you say: The local api allows up to 2.5 secs for polling but in the configuration min value is set to 10 secs?

michael-slx, 2021-12-16:
**Regarding the historical data:** Yes, it is obviously stored in WeeWX's database. However, the WeatherLink Live also has it's own internal data logger, e.g. for when the power goes out (it can run on batteries). There is currently no easy way to read that internal storage.

**2.5 second polling:** Only wind and rain sensors are supported for that 2.5 seconds-API. It works by automatically sending out that data every 2.5 seconds, whereas the rest of the sensors (temperature, humidity, etc.) have to be polled by the driver. That's where the interval can be configured. I've personally had issues with intervals below 10 seconds, e.g. the WeatherLink got stuck or acted weird. Additionally, there is little to gain with faster polling anyway since all sensors except wind and rain update only every 15 - 30 seconds.

pavll, 2021-12-16:
Ok, thank you for the quick response.

closed 2021-12-16 by pavll

pavll, 2021-12-20:
I have one last (hopefully) question: How does this work if I want to record the fastest wind speed, for example? Because if I only query data every 3 minutes, I probably won't measure exactly when the speed was fastest? And are there any options that in my web interface I update the data every 15 seconds, but only store it in the DB every 3 minutes unless it is for example the fastest wind speed or the largest amount of water I have so far?

michael-slx, 2021-12-20:
The driver itself only has the `polling_interval` option and it's usually best to leave that at the default 10 seconds.

To control how often the data is stored in the database, use WeeWX's `archive_interval` option. The high/low/sum values are then automatically calculated and stored in the database. Take a look at WeeWX documentation for more info.

Note that WeeWX by itself doesn't do anything with the 2.5 second push updates except calculate the summary every archive interval. You might want to look into third-party plugins and skins that do make use of that data.

pavll, 2021-12-20:
Ok, thanks. That's what I was searching for.  Ok and if between the 10 secs of the Weatherlink Live polling one string wind speed of 60 km/h, for example, take place but a few secs later when I poll the wind speed is back on 40 km/h. The WeeWX will not recognize 60 km/h peak as max and store it along?

michael-slx, 2021-12-22:
> [...] if between the 10 secs of the Weatherlink Live polling one strong wind speed of 60 km/h, for example, take place but a few secs later when I poll the wind speed is back on 40 km/h. The WeeWX will not recognize 60 km/h peak as max and store it along?

Yes, the maximum wind speed will be recognized. The sensor itself sends a new value every 2.5 seconds and that value is immediately sent to WeeWX, so no wind (or rain) values are lost.

> [...] how can i refresh the page more often than I want the software to data into DB?

That's not possible with WeeWX.

pavll, 2021-12-22:
Ok, thank you for all the information. So for making my website live refresh like on https://belchertownweather.com/ I would need to make a plugin that makes use of the 2.5-second push values and then refresh the report without triggering an archive action.

michael-slx, 2021-12-22:
Yes, kind of.

Regenerating the report every 2.5 seconds and uploading it would be too much. Instead, this site uses the WeeWX MQTT plugin and a separate MQTT/Websockets server to forward the data.

The weewx-belchertown GitHub repository describes this very well: https://github.com/poblabs/weewx-belchertown#mqtt-and-mqtt-websockets-optional

paulober, 2022-01-14:
How can i get the telemetry from the driver containing battery and signal strength etc... ?

michael-slx, 2022-01-14:
Examples for battery are shown in the README (`battery` mapper), look at the mapping examples.

Signal strength is not supported by the WeatherLink Live.