michael-slx/weewx-weatherlink-live #18: Barometric pressure is not plotted in weewx HTML pages

kanudude, 2021-12-30:
To properly plot barometric pressure in weewx HTML template the following changes are needed in
static/targets.py

change from:
BARO_ABSOLUTE = ["pressure"] 

change to:
BARO_ABSOLUTE = ["barometer"]

michael-slx, 2021-12-31:
To my knowledge, that would change semantics from "absolute pressure measured by the station" to "Sea Level Pressure" ([quoted from WeeWX documentation](http://www.weewx.com/docs/customizing.htm#genLoopPackets)).
Do you have any confirmation that the `bar_absolute` value returned by the WLL actually corresponds to the latter?

What is the issue with the plot?

nickbp, 2022-03-01:
If it helps, I see both sea-level and absolute pressure when manually querying `/v1/current_conditions` on a weatherlink live that's getting data from a vantage vue:
```
      {
        "lsid": xxx,
        "data_structure_type": 3,
        "bar_sea_level": 30.36,
        "bar_trend": 0.062,
        "bar_absolute": 30.253
      }
```

closed 2022-03-02 by michael-slx