michael-slx/weewx-weatherlink-live #23: Adding extra temperature sensor

NickB1, 2022-08-28:
Hi,

I added a Davis 6332 connected to a water temperature sensor to my setup.
Any idea how I can map this temperature to an extra temperature field in Weewx?

This is the data from the extra sensor transmitter:
`{"lsid":491951,"data_structure_type":1,"txid":2,"temp": 72.3,"hum":null,"dew_point":null,"wet_bulb":null,"heat_index":null,"wind_chill":null,"thw_index":null,"thsw_index":null,"wind_speed_last":null,"wind_dir_last":null,"wind_speed_avg_last_1_min":null,"wind_dir_scalar_avg_last_1_min":null,"wind_speed_avg_last_2_min":null,"wind_dir_scalar_avg_last_2_min":null,"wind_speed_hi_last_2_min":null,"wind_dir_at_hi_speed_last_2_min":null,"wind_speed_avg_last_10_min":null,"wind_dir_scalar_avg_last_10_min":null,"wind_speed_hi_last_10_min":null,"wind_dir_at_hi_speed_last_10_min":null,"rain_size":1,"rain_rate_last":0,"rain_rate_hi":0,"rainfall_last_15_min":0,"rain_rate_hi_last_15_min":0,"rainfall_last_60_min":0,"rainfall_last_24_hr":0,"rain_storm":null,"rain_storm_start_at":null,"solar_rad":null,"uv_index":null,"rx_state":0,"trans_battery_flag":0,"rainfall_daily":0,"rainfall_monthly":0,"rainfall_year":0,"rain_storm_last":null,"rain_storm_last_start_at":null,"rain_storm_last_end_at":null}`

I tried adding "soil_temp:2:1" to my config but get the following error when trying to start weewx:
`Aug 28 22:31:59 weerstation weewx[31013] DEBUG user.weatherlink_live.data_host: Polled current conditions
Aug 28 22:31:59 weerstation weewx[31013] DEBUG user.weatherlink_live.mappers: THMapping[['1']]: Mapped: outTemp=64.3
Aug 28 22:31:59 weerstation weewx[31013] DEBUG user.weatherlink_live.mappers: THMapping[['1']]: Mapped: outHumidity=61.8
Aug 28 22:31:59 weerstation weewx[31013] DEBUG user.weatherlink_live.mappers: THMapping[['1']]: Mapped: dewpoint=50.9
Aug 28 22:31:59 weerstation weewx[31013] DEBUG user.weatherlink_live.mappers: THMapping[['1']]: Mapped: heatindex=63.5
Aug 28 22:31:59 weerstation weewx[31013] DEBUG user.weatherlink_live.mappers: THMapping[['1']]: Mapped: wetbulb=55.3
Aug 28 22:31:59 weerstation weewx[31013] DEBUG user.weatherlink_live.mappers: THIndoorMapping[[]]: Mapped: inTemp=86.7
Aug 28 22:31:59 weerstation weewx[31013] DEBUG user.weatherlink_live.mappers: THIndoorMapping[[]]: Mapped: inHumidity=31.1
Aug 28 22:31:59 weerstation weewx[31013] DEBUG user.weatherlink_live.mappers: THIndoorMapping[[]]: Mapped: inDewpoint=52.6
Aug 28 22:31:59 weerstation weewx[31013] DEBUG user.weatherlink_live.mappers: THIndoorMapping[[]]: Mapped: inHeatindex=84.9
Aug 28 22:31:59 weerstation weewx[31013] DEBUG user.weatherlink_live.mappers: BaroMapping[[]]: Mapped: pressure=30.071
Aug 28 22:31:59 weerstation weewx[31013] DEBUG user.weatherlink_live.mappers: BaroMapping[[]]: Mapped: altimeter=30.151
Aug 28 22:31:59 weerstation weewx[31013] DEBUG user.weatherlink_live.mappers: RainMapping[['1']]: Mapping not responsible: Not a broadcast packet
Aug 28 22:31:59 weerstation weewx[31013] DEBUG user.weatherlink_live.mappers: WindMapping[['1']]: Mapping not responsible: Not a broadcast packet
Aug 28 22:31:59 weerstation weewx[31013] DEBUG user.weatherlink_live.mappers: SolarMapping[['1']]: Mapped: radiation=None
Aug 28 22:31:59 weerstation weewx[31013] DEBUG user.weatherlink_live.mappers: UvMapping[['1']]: Mapped: UV=None
Aug 28 22:31:59 weerstation weewx[31013] DEBUG user.weatherlink_live.mappers: ThwMapping[['1']]: Mapped: thw=63.5
Aug 28 22:31:59 weerstation weewx[31013] DEBUG user.weatherlink_live.mappers: ThswMapping[['1']]: Mapped: thsw=None
Aug 28 22:31:59 weerstation weewx[31013] DEBUG user.weatherlink_live.mappers: WindChillMapping[['1']]: Mapped: windchill=64.3
Aug 28 22:31:59 weerstation weewx[31013] ERROR user.weatherlink_live.scheduler: Error caught in scheduler tick. Not rescheduling
Aug 28 22:31:59 weerstation weewx[31013] INFO weewx.engine: Main loop exiting. Shutting engine down.
Aug 28 22:31:59 weerstation weewx[31013] DEBUG weewx.restx: Shut down Windy thread.
Aug 28 22:31:59 weerstation weewx[31013] DEBUG weewx.restx: Shut down WindGuru thread.
Aug 28 22:31:59 weerstation weewx[31013] DEBUG user.weatherlink_live.scheduler: Cancelling scheduler
Aug 28 22:31:59 weerstation weewx[31013] INFO user.weatherlink_live.scheduler: All tasks cancelled
Aug 28 22:31:59 weerstation weewx[31013] CRITICAL __main__: Caught WeeWxIOError: Error while receiving or processing packets: unsupported format character '{' (0x7b) at index 6
Aug 28 22:31:59 weerstation weewx[31013] CRITICAL __main__:     ****  Waiting 60 seconds then retrying...
`

michael-slx, 2022-08-29:
Hi,

you'll need to use the `t` mapping since it's a normal sensor transmitter, not a soil/leaf one.

I.e. the mapping would look like this: `t:2`.
The value will be mapped to one of the `extraTemp` fields. If it's your only temperature sensor apart from the main one, it should be `extraTemp1`.

NickB1, 2022-08-29:
Thanks for the suggestions, I tried the `t:2` mapping but now I get the following error: [log.txt](https://github.com/michael-slx/weewx-weatherlink-live/files/9443566/log.txt)

michael-slx, 2022-08-29:
Unfortunately I can't reproduce your issue. Can you send a log of the WeeWX initialization (i.e. the lines where it creates the mappers) and your mapping configuration line?

closed 2023-04-01 by michael-slx