michael-slx/weewx-weatherlink-live #25: Install/Update 1.1.0-2 Fails on WeeWX 5.0.0a24

gary-hammer, 2023-04-01:
I gave the new release a try as the last release (1.0.11) was working well in WeeWX 5 alphas.
I'm attaching the log from the attempt as wx5.txt

[wx5.txt](https://github.com/michael-slx/weewx-weatherlink-live/files/11129339/wx5.txt)

michael-slx, 2023-04-01:
I didn't know there were v5 alphas. I'll have to look into it.

gary-hammer, 2023-04-01:
I must have gotten lucky then with the last release.

michael-slx, 2023-04-07:
I've just pushed a new commit updating the VM scripts.

I've tested the driver with v5 alpha and it seems to work now.

gary-hammer, 2023-04-07:
Which alpha and driver version?

michael-slx, 2023-04-07:
Driver v1.1, WeeWX directly from Github repo V5 branch

gary-hammer, 2023-04-08:
Hmmm, fails to run here. It did install however.
Installing 1..0.11 works fine.

Attaching error log.
[wx-wll.txt](https://github.com/michael-slx/weewx-weatherlink-live/files/11183105/wx-wll.txt)

gary-hammer, 2023-04-19:
It looks like it fails when there is an old driver installed.

G400, 2023-12-22:
Current release (1.1.1) fails when installing in WeeWX  5.0.0b17

sudo weectl extension install weewx-weatherlink-live-v1.1.1.tar.xz
Request to install 'weewx-weatherlink-live-v1.1.1.tar.xz'.
Extracting from tar archive weewx-weatherlink-live-v1.1.1.tar.xz
Traceback (most recent call last):
  File "/usr/share/weewx/weectl.py", line 74, in <module>
    main()
  File "/usr/share/weewx/weectl.py", line 66, in main
    namespace.func(namespace)
  File "/usr/share/weewx/weectllib/extension_cmd.py", line 112, in install_extension
    ext.install_extension(namespace.source)
  File "/usr/share/weewx/weecfg/extension.py", line 122, in install_extension
    extension_name = self._install_from_file(extension_path, filetype)
  File "/usr/share/weewx/weecfg/extension.py", line 154, in _install_from_file
    extension_name = self.install_from_dir(extension_dir)
  File "/usr/share/weewx/weecfg/extension.py", line 201, in install_from_dir
    save_config |= self._inject_config(installer['config'], extension_name)
KeyError: 'config'

JPAustinTX, 2024-02-11:
Release 1.1.2 following the install instructions does not install. Weewx seems to have changed many files and seems to be broken.
> weectl extension list
Using configuration file /etc/weewx/weewx.conf
Extension Name    Version   Description
weatherlink-live  1.1.2     WeeWX driver for Davis WeatherLink Live.

So I think it's installed. 
 
> weectl station reconfigure
Only shows the default configurations. Not yours. Assuming the documentation is all wrong now and needs to be updated.

gary-hammer, 2024-02-11:
I just installed today.
All is fine with weewx 5.0.2 and 1.1.2
I used the pip install method and activated the vent
You don't mention if you are upgrading or what OS.

closed 2024-03-01 by gary-hammer