michael-slx/weewx-weatherlink-live #34: weewx 5 can not activate driver

puckthefly, 2024-01-08:
Hi,
I installed weewx5 via pip and then tried to install the weatherlink-live driver. Unfortunately this doesn't work or I'm doing something wrong, but what?

I install the driver with the command:
weectl extension install https://github.com/michael-slx/weewx-weatherlink-live/releases/download/v1.1.2/weewx-weatherlink-live-v1.1.2.tar.xz

And then I call this command:
weekly station reconfigure

Then I get the following in Driver:
Installed drivers include:
   0) ? (user.weatherlink_live) No module named 'requests'
   1) ? (user.weatherlink_live_driver) No module named 'requests'
   2) AcuRite (weewx.drivers.acurite)
   3) CC3000 (weewx.drivers.cc3000)
   4) FineOffsetUSB (weewx.drivers.fousb)
...

Now when I try to select driver 0, I get the following error message (the error message is similar for driver 1):

choose a driver [5]: 0
Traceback (most recent call last):
   File "/root/weewx-venv/bin/weekl", line 8, in <module>
     sys.exit(main())
              ^^^^^^
   File "/root/weewx-venv/lib/python3.11/site-packages/weekl.py", line 66, in main
     namespace.func(namespace)
   File "/root/weewx-venv/lib/python3.11/site-packages/weekllib/__init__.py", line 121, in dispatch
     namespace.action_func(config_dict, namespace)
   File "/root/weewx-venv/lib/python3.11/site-packages/weekllib/station_cmd.py", line 317, in reconfigure_station
     weektllib.station_actions.station_reconfigure(config_dict=config_dict,
   File "/root/weewx-venv/lib/python3.11/site-packages/weekllib/station_actions.py", line 164, in station_reconfigure
     config_config(config_dict,
   File "/root/weewx-venv/lib/python3.11/site-packages/weekllib/station_actions.py", line 207, in config_config
     config_driver(config_dict, driver=driver, no_prompt=no_prompt)
   File "/root/weewx-venv/lib/python3.11/site-packages/weekllib/station_actions.py", line 417, in config_driver
     driver_editor, driver_name, driver_version = weecfg.load_driver_editor(final_driver)
                                                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
   File "/root/weewx-venv/lib/python3.11/site-packages/weecfg/__init__.py", line 445, in load_driver_editor
     driver_module = importlib.import_module(driver_module_name)
                     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
   File "/usr/lib/python3.11/importlib/__init__.py", line 126, in import_module
     return _bootstrap._gcd_import(name[level:], package, level)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ ^^
   File "<frozen importlib._bootstrap>", line 1206, in _gcd_import
   File "<frozen importlib._bootstrap>", line 1178, in _find_and_load
   File "<frozen importlib._bootstrap>", line 1149, in _find_and_load_unlocked
   File "<frozen importlib._bootstrap>", line 690, in _load_unlocked
   File "<frozen importlib._bootstrap_external>", line 940, in exec_module
   File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
   File "/root/weewx-data/bin/user/weatherlink_live/__init__.py", line 28, in <module>
     from user.weatherlink_live.driver import WeatherlinkLiveDriver
   File "/root/weewx-data/bin/user/weatherlink_live/driver.py", line 24, in <module>
     from user.weatherlink_live import data_host, scheduler
   File "/root/weewx-data/bin/user/weatherlink_live/data_host.py", line 29, in <module>
     from user.weatherlink_live.davis_http import start_broadcast, request_current
   File "/root/weewx-data/bin/user/weatherlink_live/davis_http.py", line 25, in <module>
     import requests
ModuleNotFoundError: No module named 'requests'

michael-slx, 2024-01-08:
This driver requires the "requests" library. You have to install it using pip or another tool before trying to configure your station.

puckthefly, 2024-01-08:
Wow, thank you!

closed 2024-01-08 by puckthefly