michael-slx/weewx-weatherlink-live #36: no wind data on id2

puckthefly, 2024-01-08:
Hello, I have a problem that I can't find a solution to.

I have an ISS on ID1 and the wind sensor is separated on ID2, so I adjusted the configuration as follows:
mapping = th:1, rain:1, wind:2, th_indoor, baro, battery:1:outTemp:rain, battery:2:wind

Unfortunately I don't get any values for wind in weewx, what am I doing wrong?

But ID2 definitely comes with values that I see live on the console and I can also see them on the Davis WeatherLink Live 6100 (http://192.168.2.18/v1/current_conditions):

{"data":{"did":"001D0A7145F8","ts":1704740156,"conditions":[{"lsid":612618,"data_structure_type":1,"**txid":1**,"temp": 22.8, "hum":88.8,"dew_point": 20.0,"wet_bulb": 21.8,"heat_index": 22.8,"wind_chill":null,"thw_index":null,"thsw_index":null,"wind_speed_last":null,"wind_dir_last ":null,"wind_speed_avg_last_1_min":null,"wind_dir_scalar_avg_last_1_min":null,"wind_speed_avg_last_2_min":null,"wind_dir_scalar_avg_last_2_min":null,"wind_speed_hi_last_2_min":null,"wind_dir_at_hi_speed_last_2_min":null,"wind_speed_avg_last_10 _min":null,"wind_dir_scalar_avg_last_10_min": null,"wind_speed_hi_last_10_min":null,"wind_dir_at_hi_speed_last_10_min":null,"rain_size":2,"rain_rate_last":0,"rain_rate_hi":0,"rainfall_last_15_min":0,"rain_rate_hi_last_15_min":0,"rainfall_last_60_min":0, "rainfall_last_24_hr":0,"rain_storm":0,"rain_storm_start_at":null,"solar_rad":null,"uv_index":null,"rx_state":0,"trans_battery_flag":0,"rainfall_daily":0,"rainfall_monthly":162,"rainfall_year":162,"rain_storm_last":162,"rain_storm_last_start_at":1704176821,"rain_storm_last_end_at":1704715261},{"lsid":612663,"data_structure_type":1,"**txid":2**,"temp ":null,"hum":null,"dew_point":null,"wet_bulb":null,"heat_index":null,"wind_chill":null,"thw_index":null,"thsw_index":null,"wind_speed_last": 4.00,"wind_dir_last":47,"wind_speed_avg_last_1_min":5.31,"wind_dir_scalar_avg_last_1_min":47,"wind_speed_avg_last_2_min":4.75,"wind_dir_scalar_avg_last_2_min":45,"wind_speed_hi_last_2_min":9.00,"wind_dir_at_ hi_speed_last_2_min":54,"wind_speed_avg_last_10_min":4.37, "wind_dir_scalar_avg_last_10_min":42,"wind_speed_hi_last_10_min":9.00,"wind_dir_at_hi_speed_last_10_min":54,"rain_size":1,"rain_rate_last":0,"rain_rate_hi":0,"rainfall_last_15_min":0,"rain_rate_hi_last_15_min":0,"rainfall_last_ 60_min ":0,"rainfall_last_24_hr":0,"rain_storm":null,"rain_storm_start_at":null,"solar_rad":null,"uv_index":null,"rx_state":0,"trans_battery_flag":0,"rainfall_daily": 0,"rainfall_monthly":0,"rainfall_year":0,"rain_storm_last":null,"rain_storm_last_start_at":null,"rain_storm_last_end_at":null},{"lsid":612617,"data_structure_type":4,"temp_in": 76.6,"hum_in":29.9,"dew_point_in": 42.8,"heat_index_in": 74.6},{"lsid":612616,"data_structure_type":3,"bar_sea_level":30.227,"bar_trend": 0.043,"bar_absolute": 28,501}]},"error":null}

michael-slx, 2024-01-08:
Please try running `netcat` as [described in the troubleshooting guide](https://github.com/michael-slx/weewx-weatherlink-live/blob/release/docs/troubleshooting.md#udp-broadcast-data). It should display a new line every 2.5 seconds. Does it do that for you?

puckthefly, 2024-01-09:
Thats what i get:

Connection received on 192.168.2.18 16876
{"did":"001D0A7145F8","ts":1704786157,"conditions":[{"lsid":612618,"data_structure_type":1,"txid":1,"wind_speed_last":null,"wind_dir_last":null,"rain_size":2,"rain_rate_last":0,"rain_15_min":0,"rain_60_min":0,"rain_24_hr":0,"rain_storm":0,"rain_storm_start_at":null,"rainfall_daily":0,"rainfall_monthly":162,"rainfall_year":162,"wind_speed_hi_last_10_min":null,"wind_dir_at_hi_speed_last_10_min":null},{"lsid":612663,"data_structure_type":1,"txid":2,"wind_speed_last":4.00,"wind_dir_last":53,"rain_size":1,"rain_rate_last":0,"rain_15_min":0,"rain_60_min":0,"rain_24_hr":0,"rain_storm":null,"rain_storm_start_at":null,"rainfall_daily":0,"rainfall_monthly":0,"rainfall_year":0,"wind_speed_hi_last_10_min":19.00,"wind_dir_at_hi_speed_last_10_min":51}]}

puckthefly, 2024-01-10:
Do you have any idea whats wrong?

michael-slx, 2024-01-10:
It will take a few days until I have time to look into it. Could you provide a WeeWX log (debug logging enabled)?

puckthefly, 2024-01-13:
Please excuse the delay. Some things were very strange and so I reinstalled the weewx5 installation several times and now it works as it should. However, I can't understand what the problem was.

I noticed a difference in the driver from weewx4 to weewx5, with weewx5 two drivers are listed as soon as you call weectl station reconfigure:
Installed drivers include:
   0) WeatherLinkLive (user.weatherlink_live)
   1) WeatherLinkLive (user.weatherlink_live_driver)
   2) AcuRite (weewx.drivers.acurite)
   3) CC3000 (weewx.drivers.cc3000)
   4) FineOffsetUSB (weewx.drivers.fousb)
   5) Simulator (weewx.drivers.simulator)
   6) TE923 (weewx.drivers.te923)
   7) Ultimeter (weewx.drivers.ultimeter)
   8) Vantage (weewx.drivers.vantage)
   9) WMR100 (weewx.drivers.wmr100)
  10) WMR300 (weewx.drivers.wmr300)
  11) WMR9x8 (weewx.drivers.wmr9x8)
  12) WS1 (weewx.drivers.ws1)
  13) WS23xx (weewx.drivers.ws23xx)
  14) WS28xx (weewx.drivers.ws28xx)

I called up driver entry 1 and it works for me. But is that true, or shouldn't there just be one entry?

michael-slx, 2024-01-13:
Interesting!

I've added entry `1` because WeeWX 4 wouldn't list the `0` entry. The entry `0` still exists for backwards compatibility with old configurations. WeeWX 5 now shows both entries.

I've just tested it with 5.0.0b17 and both seem to work fine.

Also, your `netcat` output is OK too.

closed 2024-01-14 by puckthefly