michael-slx/weewx-weatherlink-live #43: Issue with mapping: "Not a broadcast packet"

birchpoplar, 2024-04-20:
Hi,
I'm stuck with a mapping issue. Syslog excerpt below - relevant lines in bold. Has been happening for a couple weeks now, and the rain and wind charts are blank from the point in time the errors occurred. 
I was on WeeWX 4.9.1 and just upgraded to 5.0.2 via `apt`.
I had to copy over the weewx service to /etc/init.d/ manually
One possibly related factor is that when doing `sudo weectl station reconfigure` the WeatherLinkLive option does not show up under devices. So I go in manually to `weewx.conf` and edit the selection--the WWL device detail is there, I just change the selection line.
Any pointers on what is wrong here?
Thanks,
JW

Apr 20 07:52:10 sparta weewxd[385670]: DEBUG weewx.engine: Station does not support reading the time
Apr 20 07:52:10 sparta weewxd[385670]: INFO weewx.engine: Using binding 'wx_binding' to database 'weewx.sdb'
Apr 20 07:52:10 sparta weewxd[385670]: INFO weewx.manager: Starting backfill of daily summaries
Apr 20 07:52:10 sparta weewxd[385670]: INFO weewx.manager: Daily summaries up to date
Apr 20 07:52:10 sparta weewxd[385670]: INFO weewx.engine: Starting main packet loop.
Apr 20 07:52:10 sparta weewxd[385670]: DEBUG user.weatherlink_live.scheduler: Push refresh will happen every 120 scheduler ticks
Apr 20 07:52:10 sparta weewxd[385670]: DEBUG user.weatherlink_live.scheduler: Scheduler tick
Apr 20 07:52:10 sparta weewxd[385670]: DEBUG user.weatherlink_live.scheduler: Notifying poll callback
Apr 20 07:52:10 sparta weewxd[385670]: DEBUG urllib3.connectionpool: Starting new HTTP connection (1): 192.168.1.6:80
Apr 20 07:52:10 sparta weewxd[385670]: DEBUG urllib3.connectionpool: http://192.168.1.6:80 "GET /v1/current_conditions HTTP/1.1" 200 None
Apr 20 07:52:10 sparta weewxd[385670]: DEBUG user.weatherlink_live.packets: Trying to create HTTP conditions packet
Apr 20 07:52:10 sparta weewxd[385670]: DEBUG user.weatherlink_live.data_host: Polled current conditions
Apr 20 07:52:10 sparta weewxd[385670]: DEBUG user.weatherlink_live.mappers: THMapping['1']: Mapped: outTemp=49.3
Apr 20 07:52:10 sparta weewxd[385670]: DEBUG user.weatherlink_live.mappers: THMapping['1']: Mapped: outHumidity=91.4
Apr 20 07:52:10 sparta weewxd[385670]: DEBUG user.weatherlink_live.mappers: THMapping['1']: Mapped: dewpoint=46.9
Apr 20 07:52:10 sparta weewxd[385670]: DEBUG user.weatherlink_live.mappers: THMapping['1']: Mapped: heatindex=49.4
Apr 20 07:52:10 sparta weewxd[385670]: DEBUG user.weatherlink_live.mappers: THMapping['1']: Mapped: wetbulb=47.9
**Apr 20 07:52:10 sparta weewxd[385670]: DEBUG user.weatherlink_live.mappers: RainMapping['1']: Mapping not responsible: Not a broadcast packet
Apr 20 07:52:10 sparta weewxd[385670]: DEBUG user.weatherlink_live.mappers: WindMapping['1']: Mapping not responsible: Not a broadcast packet**
Apr 20 07:52:10 sparta weewxd[385670]: DEBUG user.weatherlink_live.mappers: WindChillMapping['1']: Mapped: windchill=49.3
Apr 20 07:52:10 sparta weewxd[385670]: DEBUG user.weatherlink_live.mappers: ThwMapping['1', 'appTemp']: Mapped: thw=49.4
Apr 20 07:52:10 sparta weewxd[385670]: DEBUG user.weatherlink_live.mappers: ThwMapping['1', 'appTemp']: Mapped: appTemp=49.4
Apr 20 07:52:10 sparta weewxd[385670]: DEBUG user.weatherlink_live.mappers: THIndoorMapping: Mapped: inTemp=68.2
Apr 20 07:52:10 sparta weewxd[385670]: DEBUG user.weatherlink_live.mappers: THIndoorMapping: Mapped: inHumidity=41.9
Apr 20 07:52:10 sparta weewxd[385670]: DEBUG user.weatherlink_live.mappers: THIndoorMapping: Mapped: inDewpoint=44.2
Apr 20 07:52:10 sparta weewxd[385670]: DEBUG user.weatherlink_live.mappers: THIndoorMapping: Mapped: inHeatindex=66.4
Apr 20 07:52:10 sparta weewxd[385670]: DEBUG user.weatherlink_live.mappers: BaroMapping: Mapped: pressure=29.664
Apr 20 07:52:10 sparta weewxd[385670]: DEBUG user.weatherlink_live.mappers: BaroMapping: Mapped: altimeter=29.962
Apr 20 07:52:10 sparta weewxd[385670]: DEBUG user.weatherlink_live.mappers: BatteryStatusMapping['1']: Mapped: batteryStatus1=0
Apr 20 07:52:10 sparta weewxd[385670]: DEBUG user.weatherlink_live.scheduler: Notifying push refresh callback
Apr 20 07:52:10 sparta weewxd[385670]: DEBUG user.weatherlink_live.data_host: Re-requesting UDP broadcast
Apr 20 07:52:10 sparta weewxd[385670]: DEBUG urllib3.connectionpool: Starting new HTTP connection (1): 192.168.1.6:80
Apr 20 07:52:10 sparta weewxd[385670]: DEBUG urllib3.connectionpool: http://192.168.1.6:80 "GET /v1/real_time?duration=1500 HTTP/1.1" 200 None
Apr 20 07:52:10 sparta weewxd[385670]: DEBUG user.weatherlink_live.data_host: Restarting broadcast reception
Apr 20 07:52:10 sparta weewxd[385670]: DEBUG user.weatherlink_live.davis_broadcast: Starting broadcast reception
Apr 20 07:52:10 sparta weewxd[385670]: DEBUG user.weatherlink_live.scheduler: 119 scheduler ticks until next push refresh
Apr 20 07:52:10 sparta weewxd[385670]: DEBUG user.weatherlink_live.scheduler: Next scheduler tick at 2024-04-20 11:52:20 Z
Apr 20 07:52:10 sparta weewxd[385670]: DEBUG user.weatherlink_live.driver: Entering driver loop
Apr 20 07:52:10 sparta weewxd[385670]: DEBUG user.weatherlink_live.driver: Waiting for new packet
Apr 20 07:52:10 sparta weewxd[385670]: INFO user.weatherlink_live.driver: Emitting poll packet
**Apr 20 07:52:10 sparta weewxd[385670]: DEBUG user.weatherlink_live.service: WllWindGustService: Wind observations windSpeed:windDir not in record**
Apr 20 07:52:10 sparta weewxd[385670]: INFO weewx.engine: Main loop exiting. Shutting engine down.
Apr 20 07:52:10 sparta weewxd[385670]: DEBUG weewx.manager: Daily summary version is 4.0
Apr 20 07:52:10 sparta weewxd[385670]: DEBUG user.rtcr: Shut down Thread-3 thread.
Apr 20 07:52:10 sparta weewxd[385670]: DEBUG weewx.restx: Shut down WeatherCloud thread.
Apr 20 07:52:10 sparta weewxd[385670]: DEBUG user.weatherlink_live.scheduler: Cancelling scheduler
Apr 20 07:52:10 sparta weewxd[385670]: DEBUG user.weatherlink_live.scheduler: Cancelling tick task

michael-slx, 2024-04-20:
Hi,

the messages you have highlighted are absolutely normal.

Do you have any issues with wind or rain measurements (i.e. no values are displayed)? If not, then everything is OK.

birchpoplar, 2024-04-20:
Hi - many thanks. Yes, that is the issue - I have had no Wind Speed, Wind Direction, Wind Vector or Rain total for a couple weeks now. The other graphs populate fine. See the snapshot below. Thanks, JW

[image]

michael-slx, 2024-04-20:
OK, please run the commands listed in the [**troubleshooting guide**](https://github.com/michael-slx/weewx-weatherlink-live/blob/release/docs/troubleshooting.md#manually-inspecting-data) (particularly those in the **UDP broadcast data** section) and post the output here.

birchpoplar, 2024-04-20:
So with the HTTP curl command I get the output copied in at bottom.

With `$ nc -nvulp 22222` I get no response (other than the `Bound to ...`)

I tried the `http://[your.weatherlink.live.ip]/v1/real_time?duration=1200` command (with IP replaced) and get:
`{"data":{"broadcast_port":22222,"duration":1486},"error":null}`
in the browser

**Response from HTTP**
```
> GET /v1/current_conditions HTTP/1.1
> Host: XXXXXXXXXXXXXXXXXXXX
> User-Agent: curl/7.81.0
> Accept: */*
> 
* Mark bundle as not supporting multiuse
< HTTP/1.1 200 OK
< content-type: application/json
* no chunk, no close, no size. Assume close to signal end
< 
{ [60 bytes data]
100  1300    0  1300    0     0  35936      0 --:--:-- --:--:-- --:--:-- 36111
* Closing connection 0
{
  "data": {
    "did": "XXXXXXXXXXXx",
    "ts": XXXXXXXXXXXXXXX,
    "conditions": [
      {
        "lsid": 381521,
        "data_structure_type": 1,
        "txid": 1,
        "temp": 58.9,
        "hum": 38.6,
        "dew_point": 33.7,
        "wet_bulb": 42.8,
        "heat_index": 55.9,
        "wind_chill": 58.9,
        "thw_index": 55.9,
        "thsw_index": null,
        "wind_speed_last": 2,
        "wind_dir_last": 61,
        "wind_speed_avg_last_1_min": 4.5,
        "wind_dir_scalar_avg_last_1_min": 244,
        "wind_speed_avg_last_2_min": 4.56,
        "wind_dir_scalar_avg_last_2_min": 265,
        "wind_speed_hi_last_2_min": 9,
        "wind_dir_at_hi_speed_last_2_min": 203,
        "wind_speed_avg_last_10_min": 3.81,
        "wind_dir_scalar_avg_last_10_min": 301,
        "wind_speed_hi_last_10_min": 21,
        "wind_dir_at_hi_speed_last_10_min": 340,
        "rain_size": 1,
        "rain_rate_last": 0,
        "rain_rate_hi": 0,
        "rainfall_last_15_min": 0,
        "rain_rate_hi_last_15_min": 0,
        "rainfall_last_60_min": 0,
        "rainfall_last_24_hr": 5,
        "rain_storm": 34,
        "rain_storm_start_at": 1713431220,
        "solar_rad": null,
        "uv_index": null,
        "rx_state": 1,
        "trans_battery_flag": 0,
        "rainfall_daily": 0,
        "rainfall_monthly": 332,
        "rainfall_year": 1389,
        "rain_storm_last": 17,
        "rain_storm_last_start_at": 1713126300,
        "rain_storm_last_end_at": 1713243660
      },
      {
        "lsid": 381519,
        "data_structure_type": 4,
        "temp_in": 66,
        "hum_in": 31.2,
        "dew_point_in": 34.7,
        "heat_index_in": 63.1
      },
      {
        "lsid": 381518,
        "data_structure_type": 3,
        "bar_sea_level": 29.865,
        "bar_trend": -0.041,
        "bar_absolute": 29.568
      }
    ]
  },
  "error": null
}
```

birchpoplar, 2024-04-20:
Just figured it - the config I set up a couple weeks ago included `ufw` and I just `ufw disable` and I get a response on the `nc` command.

Firewall I don't think is necessary on my LAN, but if I do want to have it, then it's open on UDP 22222?

Thanks for the help here.

michael-slx, 2024-04-20:
Yep, I was about to write that it seems to be a network issue.

Closing as fixed.