michael-slx/weewx-weatherlink-live #44: Discrepancy in Precipitation Data

cmer81, 2024-06-01:
Hello,

I am experiencing an issue with the `weewx-weatherlink-live` plugin where there appears to be a significant discrepancy in the precipitation measurements reported by WeeWX compared to those displayed on WeatherLink Cloud. For example, on Thursday, May 30th, WeeWX recorded a precipitation rate of 4.5 mm/h, while WeatherLink Cloud showed 9.4 mm/h. Similar discrepancies have been noted on other days as well.

## Weewx
[image]

## Weatherlink
[image]

Interestingly, I also retrieve data from my WeatherLink Live via Home Assistant, and the data matches what is shown in WeatherLink Cloud, leading me to believe the issue may be stemming from WeeWX.

Technical Details:

- WeatherLink-Live Plugin Version: 1.1.4
- WeeWX Version: 5.0.2

Could you please inform me if this behavior is expected, or if it might be a configuration issue or a bug in the plugin? Any help in diagnosing and resolving this issue would be greatly appreciated.

Thank you

michael-slx, 2024-06-01:
Hello!

Please try to change the `rainRate = prefer_hardware` in `weewx.conf` to `hardware`.

Here's an example:

```ini
[StdWXCalculate]

    [[Calculations]]
        pressure = prefer_hardware
        altimeter = prefer_hardware
        appTemp = prefer_hardware
        barometer = prefer_hardware
        cloudbase = prefer_hardware
        dewpoint = prefer_hardware
        ET = prefer_hardware
        heatindex = prefer_hardware
        humidex = prefer_hardware
        inDewpoint = prefer_hardware
        maxSolarRad = prefer_hardware
        rainRate = hardware # <-- Set this to `hardware`
        windchill = prefer_hardware
        windrun = prefer_hardware
```

See if this helps. (It won't affect historic data though.)

cmer81, 2024-06-01:
Hello Michael,

Thank you for the guidance. I've changed the `rainRate` setting from `prefer_hardware` to `hardware` in the `weewx.conf` file as you suggested. 

I will monitor the precipitation data closely to see if this adjustment resolves the discrepancy between the WeeWX readings and those from WeatherLink live.

I appreciate your help and will keep you updated on the progress.

cmer81, 2024-06-01:
Unfortunately, there still seems to be a discrepancy in the precipitation data after a recent rain event. Here are the values I observed:

## WeeWX: 4.2 mm/h
[image]

## WeatherLink Cloud: 7.8 mm/h
[image]

## Home Assistant (WeatherLink Live): 7.8 mm/h
[image]

It appears that WeeWX is still reporting lower precipitation rates compared to WeatherLink Cloud and Home Assistant, which are consistent with each other. Do you have any further suggestions on what might be causing this issue or additional steps I could take to resolve it?

Thank you

michael-slx, 2024-06-01:
Please upload a log in debug mode at startup of WeeWX plus an additional 10 seconds or so. Additionally, if possible search in the recent log for any errors or warnings and post them here.

 Thank you.

cmer81, 2024-06-01:
Thank you for your suggestion to run WeeWX in debug mode. I have executed WeeWX in debug mode for 10 seconds, and the debug output is now available in a Gist here: [WeeWX Debug Output](https://gist.github.com/cmer81/4c359b8bef5a2fdc7313532f880803b9).

Additionally, I have made the configuration of my WeeWX setup available in another Gist, which you can view here: [WeeWX Configuration](https://gist.github.com/cmer81/e4e0d5f1966a017649cb40f90f5e3a00).

During the collection  period (19h50)  there were no errors in the WeeWX logs that could be identified as a source of the problem.

Here is also the output of the `weectl extension list` command showing the extensions currently installed:

```bash
Extension Name    Version   Description
Belchertown       1.3.1     A clean modern skin with real time streaming updates and interactive charts. Modeled after BelchertownWeather.com
mqtt              0.24      Upload weather data to MQTT server.
weatherlink-live  1.1.4     WeeWX driver for Davis WeatherLink Live.
```

Regards,

michael-slx, 2024-06-02:
OK, your log and the config look fine to me.

I forgot to ask whether you are also having problems with the rain _amount_ measurement or just the rain rate.

cmer81, 2024-06-02:
Hello,

Thank you for following up on the measurements. I’d like to clarify the observations regarding the driver change last night:

- Before 23:20: Usage of your initial driver.
- Starting from 23:40: Switched to the [weatherlinklive-driver-weewx ](https://github.com/Drealine/weatherlinklive-driver-weewx)driver.

Regarding your query about rain measurements: I am not experiencing any problems with the rain amount measurement, this value appear to be consistent.

However, I have noticed an inconsistency in the atmospheric pressure readings after switching drivers. The `weatherlinklive-driver-weewx` reports atmospheric pressure values that are similar to those seen on the WeatherLink Cloud, differing from the values provided by your initial driver.

## Weewx
[image]

[image]

## Weatherlink Cloud
[image]

I am available to provide further details.

michael-slx, 2024-06-02:
The difference in the pressure has to do with the different types of pressure measurements WeeWX supports. The WeatherLink documentation isn't clear on what "kind" of pressure value it reports.

I will have to further look into the rain rate discrepancies.

cmer81, 2024-06-03:
Hello,

Thank you for your suggestion. I have set up a test secondary instance of WeeWX using the `weewx-weatherlink-live` and have incorporated the modification to the `Accumulator` as you mentioned.

```ini
[accumulator]
  [[rainRate]]
    extractor = max
```   

Now, I just need to wait for the next rain, which is expected by the end of this week, to see if this resolves the issue.

Thanks again for your help.

michael-slx, 2024-06-21:
Has the configuration change solved your problem?

closed 2024-06-22 by michael-slx

cmer81, 2024-06-29:
Sorry for my late response, I was on vacation.
Yes, I confirm that it improves the situation and accurately reflects the WeatherLink data.

Thanks for your help 🙇

commit dbdc4bb6 (2024-06-22): Set `max` accumulator for `rainRate` metric

Rain rate is accumulated by using maximum value for consistency
with WeatherLink app/dashboard.

Fixes #44